rpng/calc #2: what is the dataset you use in the ros online demo

zqnnn, 2018-05-05:
what is the pos_topic

nmerrill67, 2018-05-05:
Good question. I made a custom publisher for KITTI data. There is a python script that does it [here](https://github.com/nmerrill67/kitti_publisher). The EuRO MAV datasets also work out of the box with that demo, and I also have [another python script](https://github.com/nmerrill67/new_college_publisher) to publish the New College Dataset in the right format. The online demo needs an Image topic as well as a topic with trajectories either as a TransformStamped or PoseStamped topic.

zqnnn, 2018-05-17:
Thanks! Another question,please. What is the training dataset used by the calc caffemodel?

nmerrill67, 2018-05-22:
We use the MIT Places dataset. Please see http://arxiv.org/abs/1805.07703

closed 2018-05-22 by nmerrill67