rpng/calc #4: A standalone sample

mpkuse, 2018-07-03:
Hi, 
Can you suggest me how can I test your algorithm with my own sequence? So my sequence is a series of images very similar to gradens-point dataset. I wish to extract the scene descriptors from your method for each of the images and compare the loop detections.

nmerrill67, 2018-07-03:
Hi Manohar,

For general documentation, I would go to TrainAndTest/main.py and run it
with the args: ' test -h'. Basically, you can create precision recall
curves with your dataset if you place the query and database images in
directories called live and memory, respectively. Then analyze the
sequential overlap for your dataset (i.e., multiple database matches i-n/2,
..., i, ..., i+n/2 for query image i in order to choose the parameter n.
Finally, choose the right arguments to choose the aforementioned parameters
based on the help command as mentioned, and point to the model you wish to
test.

mpkuse, 2018-07-03:
So, I downloaded your caffe models in `DeepLCD/calc_model`. Also untared your CampusLoopDataset. 

Finally running, 
```
python main.py test -m ../DeepLCD/calc_model/ -d test_data/CampusLoopDataset/

Traceback (most recent call last):
  File "main.py", line 157, in <module>
    plot(net_model_path=args.model_path, data_path=args.data_path, num_include=args.num_include, title=args.title, resize_net=args.rz_net, alexnet_proto_path=args.alex_proto, alexnet_weights=args.alex_weights)
  File "/home/mpkuse/Downloads/calc/TrainAndTest/testNet.py", line 326, in plot
    precisions, recalls, precision_alex, recall_alex, precision_hog, recall_hog, precision_dbow, recall_dbow = get_prec_recall(net_def_path, net_model_path, data_path, num_include, title, resize_net, alexnet_proto_path, alexnet_weights)
  File "/home/mpkuse/Downloads/calc/TrainAndTest/testNet.py", line 131, in get_prec_recall
    nets.append(caffe.Net(net_def_path,1,weights=m_path))
RuntimeError: Could not open file proto/deploy.prototxt
```

I can see it cannot find the prototxt file, Am I specifying the arguments wrongly? What is the correct way?

nmerrill67, 2018-07-03:
Right,

The code expects the CALC prototxt model definition file to be at that
location listed in your error. A quick fix is to copy it there, but I need
to add an arg for that.

mpkuse, 2018-07-03:
Ok! Thanks got it working !

mpkuse, 2018-07-04:
I am trying to get the descriptors for my images. I am trying to do something similar to this: https://github.com/rpng/calc/blob/166f90c4e77637fc030911bc4f65a79bcaa2b0b8/TrainAndTest/testNet.py#L446 ie. your `function view_forward_pass()`. 

Can you confirm that `net.blobs['descriptor'].data` gets me the descriptor for my image using your proposed auto-encoder model?

nmerrill67, 2018-07-04:
Can you specify exactly what you need to do? I have that function for
viewing the data as a heat map as in our paper. I also have the entire C++
library in DeepLCD for performing image matching. If you want to know how
to do it for your own python code then the function in testNet is a good
example. The descriptor layer is in fact the layer for the descriptor.

nmerrill67, 2018-07-11:
I forgot to answer the last part of your question sorry. Yes that model from BVLC is the model we used for AlexNet in the paper.

mpkuse, 2018-07-12:
Ok. Thanks for all your replies. I got your code working as I desired.

closed 2018-07-12 by nmerrill67