rpng/calc #5: An error in TrainAndTest

zqnnn, 2018-07-05:
Excuse me. When I run the script
python main.py test -m ../DeepLCD/calc_model/ -d test_data/CampusLoopDataset/ 
I get this error. Can you help me? 

[image]

zqnnn, 2018-07-05:
I0705 16:01:36.815151 14483 net.cpp:137] Memory required for data: 5063648
I0705 16:01:36.815152 14483 net.cpp:200] descriptor does not need backward computation.
I0705 16:01:36.815155 14483 net.cpp:200] relu3 does not need backward computation.
I0705 16:01:36.815158 14483 net.cpp:200] conv3 does not need backward computation.
I0705 16:01:36.815160 14483 net.cpp:200] norm2 does not need backward computation.
I0705 16:01:36.815163 14483 net.cpp:200] pool2 does not need backward computation.
I0705 16:01:36.815166 14483 net.cpp:200] relu2 does not need backward computation.
I0705 16:01:36.815170 14483 net.cpp:200] conv2 does not need backward computation.
I0705 16:01:36.815171 14483 net.cpp:200] norm1 does not need backward computation.
I0705 16:01:36.815173 14483 net.cpp:200] pool1 does not need backward computation.
I0705 16:01:36.815176 14483 net.cpp:200] relu1 does not need backward computation.
I0705 16:01:36.815178 14483 net.cpp:200] conv1 does not need backward computation.
I0705 16:01:36.815181 14483 net.cpp:200] X1 does not need backward computation.
I0705 16:01:36.815182 14483 net.cpp:242] This network produces output descriptor
I0705 16:01:36.815191 14483 net.cpp:255] Network initialization done.
HDF5-DIAG: Error detected in HDF5 (1.8.16) thread 139991413573376:
  #000: ../../../src/H5F.c line 439 in H5Fis_hdf5(): unable open file
    major: File accessibilty
    minor: Not an HDF5 file
  #001: ../../../src/H5Fint.c line 558 in H5F_is_hdf5(): unable to locate file signature
    major: File accessibilty
    minor: Not an HDF5 file
  #002: ../../../src/H5FDint.c line 146 in H5FD_locate_signature(): unable to read file signature
    major: Low-level I/O
    minor: Unable to initialize object
  #003: ../../../src/H5FDint.c line 211 in H5FD_read(): driver read request failed
    major: Virtual File Layer
    minor: Read failed
  #004: ../../../src/H5FDsec2.c line 707 in H5FD_sec2_read(): file read failed: time = Thu Jul  5 16:01:36 2018
, filename = '/home/zqn/project/calc-new/DeepLCD/calc_model/', file descriptor = 25, errno = 21, error message = 'Is a directory', buf = 0x7ffddddf5650, total read size = 8, bytes this sub-read = 8, bytes actually read = 18446744073709551615, offset = 0
    major: Low-level I/O
    minor: Read failed
HDF5-DIAG: Error detected in HDF5 (1.8.16) thread 139991413573376:
  #000: ../../../src/H5F.c line 604 in H5Fopen(): unable to open file
    major: File accessibilty
    minor: Unable to open file
  #001: ../../../src/H5Fint.c line 1087 in H5F_open(): unable to read superblock
    major: File accessibilty
    minor: Read failed
  #002: ../../../src/H5Fsuper.c line 275 in H5F_super_read(): unable to locate file signature
    major: File accessibilty
    minor: Not an HDF5 file
  #003: ../../../src/H5FDint.c line 146 in H5FD_locate_signature(): unable to read file signature
    major: Low-level I/O
    minor: Unable to initialize object
  #004: ../../../src/H5FDint.c line 211 in H5FD_read(): driver read request failed
    major: Virtual File Layer
    minor: Read failed
  #005: ../../../src/H5FDsec2.c line 707 in H5FD_sec2_read(): file read failed: time = Thu Jul  5 16:01:36 2018
, filename = '/home/zqn/project/calc-new/DeepLCD/calc_model/', file descriptor = 25, errno = 21, error message = 'Is a directory', buf = 0x7ffddddf52a0, total read size = 8, bytes this sub-read = 8, bytes actually read = 18446744073709551615, offset = 0
    major: Low-level I/O
    minor: Read failed
F0705 16:01:36.815745 14483 net.cpp:791] Check failed: file_hid >= 0 (-1 vs. 0) Couldn't open /home/zqn/project/calc-new/DeepLCD/calc_model/

nmerrill67, 2018-07-05:
Hi,

I think you need to run the 'get_model' script in DeepLCD to download our model. Then it should work.

zqnnn, 2018-07-06:
But I have already run this command， there is the calc model in the folder

goldbattle, 2018-07-06:
From the looking more closely at your error message, it is trying to open the caffe model.
You should checkout what the default is:
https://github.com/rpng/calc/blob/166f90c4e77637fc030911bc4f65a79bcaa2b0b8/TrainAndTest/main.py#L106

You need to specify the *file*, not the directory.
Please try directly specifying the model like the default argument.
If it still errors, you could have a corrupted download, so you can always try redownloading the model.

closed 2018-07-07 by goldbattle