rpng/calc #6: Question about training on my own dataset

zqnnn, 2018-07-10:
Hello, if I want  to use your net model train on my dataset, what should I do?

nmerrill67, 2018-07-11:
Hello,

If I am understanding correctly, you want to fine tune our pretrained model with your own dataset?

If that's the case, then first create the LMDBs for your dataset. You can see how to do this by running `./main.py db -h`. Make sure you don't specify the `--train-after` flag. After those are made, you can run `./main.py net --define --train -w model/calc.caffemodel -x1 <your X1> -x2 <your X2>` to define and train the model, initializing the weights to the previous model. Simply skip the `-w model/calc.caffemodel` to train from scratch. You can change the learning rate in `makeNet.py` in the line `self.sp['base_lr'] = '0.0009' #'0.0018'` if you want to lower it for fine tuning.

zqnnn, 2018-07-12:
Thank you. I have read your paper. I don't know why you say that "Without any extra optimization constraints, the network  learned  zero  vectors". I want to reconstruct the raw image.
[image]

nmerrill67, 2018-07-12:
Right. I think we should take this conversation off the issues feed. Please email me at nmerrill@udel.edu with questions about the paper specifically.

nmerrill67, 2018-07-12:
Closing for now since this is not specifically related to the repo any more.

closed 2018-07-12 by nmerrill67