rpng/calc #14: 100 landmarks

Chatoyant19, 2019-03-04:
Hello, thank you for sharing your code. How did you get the 100 landmarks? First get some proposals using BING, and then generate the final detection using a detection? But the results of the final detection are some bounding boxes. I think landmarks should be some points rather than bounding boxes. Thank you very much.

nmerrill67, 2019-03-11:
Typically in SLAM, yes, landmarks are points, but in [this paper](https://eprints.qut.edu.au/84931/1/rss15_placeRec.pdf) that we were reproducing with CALC, landmarks are bounding boxes of buildings and other prevalent objects. I would suggest reading the linked paper (cited in our paper as well) to get a better picture of what is going on.

Chatoyant19, 2019-03-12:
Thanks for your response.I train the net from the scratch on the Places dataset(256*256), and I test the model on the campus dataset(your paper).I set the maximum number of iterations to 500k and use iteration 220k(calc_iter_220000.caffemodel) to test the testing dataset as you said in other issues, I also used other iterations to get the caffemodel to test the testing dataset. But my results are much worse than yours. I use the default batch size 256, and a single GPU(GeForce GTX 1080) for training.Can you tell me how to get experimental results similar to yours?

nmerrill67, 2019-03-14:
Please see previous issues that discuss this issue. I used a larger batch size with two GPUs. The default is set to 256 for convenience for people who may not have the 12GB GPUs.

closed 2019-03-14 by nmerrill67

Mingrui-Yu, 2020-04-17:
> experimental

hello~ Have you trained the net successfully? I'm also interested in retraining the net and I think I'm facing the same problem.

Mingrui-Yu, 2020-04-17:
@Chatoyant19
Hello~ Have you trained the net successfully? I'm also interested in retraining the net and I think I'm facing the same problem.