rpng/calc #18: make error

Leontree, 2019-04-16:
Hi, nate. When compile:

make[2]: *** No rule to make target 'src/googletest-build/googlemock/gtest/libgtest.a', needed by 'deeplcd-test'。 停止。
CMakeFiles/Makefile2:211: recipe for target 'CMakeFiles/deeplcd-test.dir/all' failed
make[1]: *** [CMakeFiles/deeplcd-test.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

Do you know how to deal with this issue??

nmerrill67, 2019-04-20:
Yes the gtest repo changed their code so that my method of compiling against the current repo fails. I do not have a good fix for this yet, but all of the code besides the unit tests will still work. If you do not want to see the error simply set the BUILD_TESTS cmake option to off when compiling for now until I can fix the unit test compiling.

Leontree, 2019-04-20:
> Yes the gtest repo changed their code so that my method of compiling against the current repo fails. I do not have a good fix for this yet, but all of the code besides the unit tests will still [...]

Thanks a lot ! Nate, have a nice day :D

nmerrill67, 2019-10-17:
I fixed it by forking an older gtest.

closed 2019-10-17 by nmerrill67