rpng/calc #21: compile DBoW2 error

JinghuiChan, 2019-09-15:
Hi Nate
    When I compiled DBoW2, I got the following error:
   calc/TrainAndTest/ThirdParty/DBoW2/src/pywrap.cpp:125:17: error: ‘boost::python::numeric’ has not been declared
  boost::python::numeric::array::set_module_and_type("numpy", "ndarray");

 How should I solve it?
[image]

nmerrill67, 2019-09-15:
Mine compiles just fine (just checked). My boost version is 1.58.0 and python is 2.7.12.
Make sure you have `libboost-python1.58-dev` installed.

JinghuiChan, 2019-09-17:
Thank you, after the boost version was changed, the compilation was successful.

closed 2019-09-17 by JinghuiChan