rpng/calc #22: Run testNet.py

JinghuiChan, 2019-09-19:
Hi Nate.
    I have installed all the required dependencies and successfully compiled the DeepLCD library. Now I want to see the precision-recall curves(comparing against HOG and DBoW2), so I run the following command:
 $ cd ~/calc/TrainAndTest
 $ python testNet.py
Unfortunately, It is nothing output.
What commands should I run to print the precision-recall curves?

AllofLife, 2019-09-19:
you can try python2 main.py -h
if you donn't want to load the calc model  you should edit the fution plot code in testnet.py  
although i am not the author,but i hope it can help you

nmerrill67, 2019-09-19:
@AllofLife is correct. main.py is the entry point to preprocess data, train, and test with PR curves.

closed 2019-09-19 by nmerrill67