rpng/calc #23: deeplcd-test error

PietroAzzari, 2019-11-07:
Hello,
i am on Ubuntu16.04 and build Caffe from src in CPU_ONLY mode.
i run ./deeplcd-test
and got this 
...
I1107 10:07:56.879565 11560 net.cpp:202] relu1 does not need backward computation.
I1107 10:07:56.879568 11560 net.cpp:202] conv1 does not need backward computation.
I1107 10:07:56.879571 11560 net.cpp:202] X1 does not need backward computation.
I1107 10:07:56.879573 11560 net.cpp:244] This network produces output descriptor
I1107 10:07:56.879580 11560 net.cpp:257] Network initialization done.
I1107 10:07:56.899730 11560 net.cpp:746] Ignoring source layer X2
I1107 10:07:56.899821 11560 net.cpp:746] Ignoring source layer fc4
I1107 10:07:56.899827 11560 net.cpp:746] Ignoring source layer sig4
I1107 10:07:56.899828 11560 net.cpp:746] Ignoring source layer fc5
I1107 10:07:56.899830 11560 net.cpp:746] Ignoring source layer sig5
I1107 10:07:56.899832 11560 net.cpp:746] Ignoring source layer fc6
I1107 10:07:56.899834 11560 net.cpp:746] Ignoring source layer X2_comp_hat
I1107 10:07:56.899837 11560 net.cpp:746] Ignoring source layer Flatten1
I1107 10:07:56.899838 11560 net.cpp:746] Ignoring source layer loss

Caffe mode = CPU
Loaded model in 0.020969 seconds
[       OK ] DeepLCDTest.TestFullConstructor (22 ms)
[ RUN      ] DeepLCDTest.TestAddMulti
[       OK ] DeepLCDTest.TestAddMulti (127 ms)
[ RUN      ] DeepLCDTest.TestMultiImageRetrieval
[       OK ] DeepLCDTest.TestMultiImageRetrieval (127 ms)
[----------] 6 tests from DeepLCDTest (575 ms total)

[----------] Global test environment tear-down
[==========] 6 tests from 1 test case ran. (575 ms total)
[  PASSED  ] 6 tests.
*** Error in `./deeplcd-test': free(): invalid next size (normal): 0x0000000002472f80 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7ff48e6417e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7ff48e64a37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7ff48e64e53c]
/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4(_ZNSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS5_EED1Ev+0x29)[0x7ff48fe72ad9]
/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0x9a)[0x7ff48e60436a]
/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4(+0x1f633)[0x7ff48fe72633]
This is only the last part of the log, the full log is attached.
Let me know if you have any suggestions.
Thanks and regards,
Pietro
[log.txt](https://github.com/rpng/calc/files/3818742/log.txt)

nmerrill67, 2019-11-19:
I'm a little confused, I do not see that error in the log you attached. Have you resolved this error?

PietroAzzari, 2019-11-21:
Hi,
sorry i redirected into the file but probably the missing part was output into stderr and that one was not redirected.   here it is the full version
[log.txt](https://github.com/rpng/calc/files/3875412/log.txt)

Thanks for helping
Pietro

closed 2020-07-10 by nmerrill67