rpng/calc #24: make error

YDY-andy, 2019-11-11:
[ 50%] Building CXX object CMakeFiles/deeplcd.dir/src/deeplcd/deeplcd.cpp.o
In file included from /home/ydy/calc/DeepLCD/src/deeplcd/deeplcd.cpp:1:0:
/home/ydy/calc/DeepLCD/include/deeplcd/deeplcd.h:1:27: fatal error: caffe/caffe.hpp: no such file or dictionary
compilation terminated.
CMakeFiles/deeplcd.dir/build.make:62: recipe for target 'CMakeFiles/deeplcd.dir/src/deeplcd/deeplcd.cpp.o' failed
make[2]: *** [CMakeFiles/deeplcd.dir/src/deeplcd/deeplcd.cpp.o] Error 1
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/deeplcd.dir/all' failed
make[1]: *** [CMakeFiles/deeplcd.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

I am using Ubuntu16.04 and I'm sure I install caffe under /home/usr. I don't know Y it can not find caffe.hpp, it acually exist under  /home/usr/caffe/include/caffe

YDY-andy, 2019-11-11:
@nmerrill67 I finally found that in the cmakelist of DeepLCD
set(Caffe_INCLUDE_DIRS ${Caffe_ROOT_DIR}/build/install/include)  may be  wrong

in my caffe pack there is no such folder ,so I change it to set(Caffe_INCLUDE_DIRS ${Caffe_ROOT_DIR}/include),then it worked

of cource there's still error about no caffe/proto/caffe.pb.h ,but this is easy to be solved

nmerrill67, 2019-11-19:
Did you change the install path for Caffe? There is also a FindCaffe cmake file that I may try out. This method has been working for me on multiple machines though.

closed 2020-07-10 by nmerrill67

PietroAzzari, 2020-12-09:
HI,
i had the same issue until i actually run make install. At that point you will find the headers at ${Caffe_ROOT_DIR}/build/install/include as specified in the cmakelists file