rpng/calc #26: What is the value of loss?

wangtiandong-roborock, 2020-04-01:
Hi @nmerrill67 ,
I'm finetuing calc network on our own dataset. Could you provide the final value of your training loss? Mine is 63, which I think is too large, and the PR Curve is also worse than your weights.
I trained the network 500k iterations.

closed 2020-12-02 by nmerrill67

syqlyx, 2021-04-23:
> Hi @nmerrill67 ,
> I'm finetuing calc network on our own dataset. Could you provide the final value of your training loss? Mine is 63, which I think is too large, and the PR Curve is also worse than your weights.
> I trained the network 500k iterations.

Excuse me, have you solved this problem? I have a similar problem now

wangtiandong-roborock, 2021-04-23:
> Excuse me, have you solved this problem? I have a similar problem now

No, I haven't. I found there were lots of other methods better than calc network. I do suggest you read more recent papers.

syqlyx, 2021-04-23:
> No, I haven't. I found there were lots of other methods better than calc network. I do suggest you read more recent papers.

Got it! Thanks for your suggestion~