rpng/calc #27: Using DeepLCD in LSD-SLAM

WuSa12956, 2020-07-10:
Hi, I have now embedded Deeplcd into LSD-SLAM for closed-loop detection.  But when I run ROS, I get the error in the picture.
[image]

nmerrill67, 2020-07-10:
Cool project! If you look at my example launch file `online-demo.launch`, you will see the `net_def` and `weights` params. It looks like yours are not pointing to the right location. Make sure they point to where you have the .prototxt and .caffemodel files.

WuSa12956, 2020-07-10:
@nmerrill67 
Hello, I use DeepLCD to filter candidate frames.  The frame with the highest score is selected as the candidate frame.  After that, close-loop detection is performed.  In this case, do I need the online-demo example to improve my code?  

The error in the picture is the error that the command ‘rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info’ appeared after I added the data set bag file.  Does DeepLCD need to be compiled separately before use?

This is the code I used to refer to the.prototxt and.caffemodel file。
[image]

Many thanks

nmerrill67, 2020-07-10:
No you don't need to compile it separately. You just need to make sure you set the arguments for DeepLCD to the correct file locations wherever you instantiate the DeepLCD object. You can just look at my online demo for how to do this. You will just need to make it a ros::NodeHandle param. Making a launch file will make this easier -- especially to set other params like the GPU and whatnot. Are you familiar with how to do this with ROS? LSD SLAM has it a bit convoluted compared to normal usage, but the node handle is in `IOWrapper/ROS/ROSImageStreamThread.h`, where you can potentially set the param like I do [here](https://github.com/rpng/calc/blob/e458126799a647bacdfd233c9bec8952453d720d/DeepLCD/online-demo_ws/src/online-demo.cpp#L116)

closed 2020-07-20 by WuSa12956