rpng/calc #28: When I execute cmake .. && make in the build folder of DeepLCD I get the following error

Skcoica, 2020-09-03:
I installed everything laid out in the README.md and even tried installing gtest manually, but this persists.

[image]

nmerrill67, 2020-09-03:
It seems that you are not building it from within the build directory. This could at least explain the `cp` errors. You should run `cmake ..` from within `DeepLCD/build`.

Skcoica, 2020-09-03:
My bad, it seems I have uploaded the wrong Screenshot
[image]

nmerrill67, 2020-09-03:
Do you have git installed? It seems like that is (one of) the error(s) there.

closed 2020-12-02 by nmerrill67