rpng/calc #29: make error

HTaCheng, 2020-11-27:
Hello, how can I solve this problem?
[image]

nmerrill67, 2020-11-30:
You must point [this cmake variable](https://github.com/rpng/calc/blob/e458126799a647bacdfd233c9bec8952453d720d/DeepLCD/CMakeLists.txt#L28) to the install path of caffe.

HTaCheng, 2020-12-02:
Thank you very much, I found a solution. But I encountered a problem again when cmake .. && make DBoW2. Please help to see how to solve it?
[image]

nmerrill67, 2020-12-02:
I get the same error now. It looks like a dependency of DBow2 is now incompatible with the new OpenCV. I can take a look later this week.

HTaCheng, 2020-12-02:
Hello,can you tell me the version of opencv you used at the time,and I will try to compile it.

HTaCheng, 2020-12-02:
I am using opencv 3.4.8.

nmerrill67, 2020-12-02:
Based on the [last travis build](https://travis-ci.org/github/rpng/calc) it looks like OpenCV 2.4.8. That version is super outdated at this point but if you want to try it for now it should work. As I said, I am a bit busy now, but I can try to make it compatible with newer OpenCV later this week.

HTaCheng, 2020-12-02:
Okay, you are busy with your work first. Looking forwoard to your reply.

HTaCheng, 2020-12-06:
Hello, the following error occurred when I was training the network, how should I download this file.
[image]

nmerrill67, 2020-12-07:
So I should have the issue fixed on [my branch of DBoW2](https://github.com/nmerrill67/DBoW2). See the README, you now have to use my branch of DLib too or else the same error will persist.

HTaCheng, 2020-12-08:
Thanks you for your kind help, but I encountered a problem again, how to I install the python2 version of boost_numpy.
[image]
[image]