rpng/calc #32: No Image received

chenjiangtao229, 2021-09-07:
Hello, i've been trying your code recently, but i ran into this problem with DeepLCD/online-demo_ws. Can you help me with this?
[image]

Dhuzi96, 2021-09-15:
你好啊同学，请教一下你啊，请问源码中testNet.py中PR曲线的计算是怎么得来的你弄清楚了吗

closed 2021-09-26 by chenjiangtao229

Dhuzi96, 2021-09-26:
大佬，别忽略呀>.<

chenjiangtao229, 2021-09-26:
@Dhuzi96  我给你发邮箱了，没收到吗？ 关于上个问题：我caffe安装的是only-cpu的，目前也就没认真看，接下来应该会研究一下的。

Dhuzi96, 2021-09-26:
没收到哎，能再发一次吗？( * v *  )                      这个邮箱   clerkhu@google.com

chenjiangtao229, 2021-09-26:
发不过去啊！
PICTURE:/home/chenjiangtao/.config/tencent-qq/1606188988/image/origin/77AE25B9878CF2A816828895D5E68207.png

chenjiangtao229, 2021-09-26:
发不过去啊！
[image]

Dhuzi96, 2021-09-26:
错了错了是：clerkhu@gmail.com      不好意思不好意思